ethz-asl/ethzasl_sensor_fusion #36: form rxplot to rqt_plot ?

iamfarhan007, 2015-04-28:
Hi 
Is it possible to convert /ssf_cor/scripts files to rqt_plot version to work on ununto 14.04LTS and Indigo
Thanks
Farhan

stephanweiss, 2015-04-28:
Farhan,

Yes, this is on our todo list since quite a while. If you would be willing to do these changes and submit a pull request, that would be great.

Best,
Stephan

iamfarhan007, 2015-04-28:
Thanks Stephan
i know this is not the right form to ask but can you have look at my quarries here
https://github.com/ethz-asl/asctec_mav_framework/issues/38

its been ages since any reply.

thanks
farhan

shanmo, 2016-01-24:
It works with 

rqt_plot ssf_core/state_out/data[0]:data[1]:data[2] ssf_core/state_out/data[3]:data[4]:data[5] &
rqt_plot ssf_core/state_out/data[13]:data[14]:data[15] ssf_core/state_out/data[16]

However, I could not figure out a way to change the legend for rqt_plot

zanazakaryaie, 2017-07-16:
Hello

I'm looking for a version to be used in ROS idigo or jade. Is there any solution?

Regards

Zana