ethz-asl/ethzasl_sensor_fusion #39: Can not find "PositionWithCovarianceStamped.h"

TingruW, 2015-09-01:
Hi Stephan,
I am really fascinated by your work and want to see if it is possible to be transplanted on mobile phone.
But when I finished downloading and using catkin_make to do the compilation, this error appeared  
fatal error: ssf_updates/PositionWithCovarianceStamped.h: No such file or directory
 #include <ssf_updates/PositionWithCovarianceStamped.h>
Is there anything missing? How could I  continue the compilation.

Thanks in advance.
Tingru

stephanweiss, 2015-09-01:
This seems to be a missing message type. Please verify that you have the file PositionWithCovarianceStamped.msg in the msg folder of ssf_updates (should be in there by default, otherwise, download the latest version). If the error still occurs, this means that something is wrong with the CMakeLists.txt file or with your Catkin build environment (i.e. the messages are not built correctly: PositionWithCovarianceStamped.h will be created by the build system from PositionWithCovarianceStamped.msg).

Best,
Stephan

adamjardim, 2016-06-09:
This might be reviving the dead a bit, but I also encountered this issue, stumbled upon this thread hoping for a solution, but didn't find one here!  For those who are looking to build this package and encounter this issue, my solution was to add:

`add_dependencies(position_sensor ssf_updates_generate_messages_cpp)`

to the ssf_updates CMakelist (ethzasl_sensor_fusion/ssf_updates/CMakeLists.txt) just after the add_executable and set_property lines for position_sensor.

Should look something like this:

`add_executable(position_sensor src/main.cpp src/position_sensor.cpp)`
`set_property(TARGET position_sensor PROPERTY COMPILE_DEFINITIONS POSITION_MEAS)`
`add_dependencies(position_sensor ssf_updates_generate_messages_cpp)` <================
`target_link_libraries(position_sensor ${catkin_LIBRARIES})`

It looks like the build machines they're running probably built those files first, but a less capable machine (like mine) doesn't have the ability to work as in parallel as theirs.  As such, explicitly adding that line makes it buildable for me.

Hope this helps!