ethz-asl/ethzasl_sensor_fusion #40: Tutorial worked first time only

ButlerRobotics, 2015-09-10:
Strangest thing happened at the end of completing the getting started tutorial:
http://wiki.ros.org/ethzasl_sensor_fusion/Tutorials/getting_started

I followed the instructions and then ran the launch file. I see the graph working as expected (updated to rqt_plot). I then let the bag file run until it gets to the end. I'm thinking so far so good but then I try to run it again, now it won't work for me.

I have tried reinitialization everything again. I try over and over. The graph is now flat lining. I can't understand why it would only work once when nothing has changed.

The only thing that I can think of that may be different each time, is the time that I initialize the filter. This is always different as I stop the bag file after running it for approximately 1 second and then initialize as per instructions.

I either get an error warning: 
=== ERROR ===  update: NAN at index 0
=== ERROR ===  prediction p: NAN at index 0

Or I get this message: 
[ WARN] [1441884003.214357239]: large time-gap re-initializing to last state

And sometimes this:
[ WARN] [1441884003.679369485]: Negative scale detected: -0.39217. Correcting to 0.1

It is different depending on where the 'play head' is when I initialize.

ButlerRobotics, 2015-09-10:
Don't know if it is relevant but I get these messages when building:

/home/karl/catkin_ws/src/ethzasl_sensor_fusion/ssf_updates/src/viconpos_sensor.cpp: In member function ‘void PositionSensorHandler::measurementCallback(const TransformStampedConstPtr&)’:
/home/karl/catkin_ws/src/ethzasl_sensor_fusion/ssf_updates/src/viconpos_sensor.cpp:116:28: warning: variable ‘C_ci’ set but not used [-Wunused-but-set-variable]
  Eigen::Matrix<double,3,3> C_ci = state_old.q_ci_.conjugate().toRotationMatrix();
                            ^
/home/karl/catkin_ws/src/ethzasl_sensor_fusion/ssf_updates/src/position_sensor.cpp: In member function ‘void PositionSensorHandler::measurementCallback(const PositionWithCovarianceStampedConstPtr&)’:
/home/karl/catkin_ws/src/ethzasl_sensor_fusion/ssf_updates/src/position_sensor.cpp:113:28: warning: variable ‘C_ci’ set but not used [-Wunused-but-set-variable]
  Eigen::Matrix<double,3,3> C_ci = state_old.q_ci_.conjugate().toRotationMatrix();

ButlerRobotics, 2015-09-12:
Also logged at ROS answers: http://answers.ros.org/question/217554/ethzasl_sensor_fusion-only-works-first-time/

robberM, 2015-10-22:
hi,ButlerRobotics ,
I met the same problem as yours, have you solved it ?

ButlerRobotics, 2015-10-22:
@robberM No sorry. Love to know what the issue is but no one seems to know. Please let me know if you figure it out.

kubiak54, 2016-01-07:
Make sure that in viconpos_sensor.cpp you commented only IF part of "// take covariance from sensor", ELSE part should not be commented.

zuomaidi52, 2016-04-15:
@ButlerRobotics  Hi,ButlerRobotics,
Now, i have the platform the same of your.(ubuntu 14.04 + indigo),but ,**I can't ran in the first time**.
The catkin_make error info: 
`/home/zuo/ws/src/ethzasl_sensor_fusion-maste/ssf_updates/src/viconpos_measurements.h:46: error: undefined reference to PositionSensorHandler::PositionSensorHandler(ssf_core::Measurements*)`

- I used 
`#ifdef POSITION_MEAS
#undef POSITION_MEAS
#endif
#ifdef POSE_MEAS
#undef POSE_MEAS
#endif`
to undefine those Macro.

- And I used absolute path to write include file
like this
'#include "/home/zuo/ws/src/ethzasl_sensor_fusion-maste/ssf_updates/src/viconpos_sensor.h"'

Now I was completely obsessed with this. Plz help me!
And my email : zuomaidi@126.com
Thank you.

ChildAgo, 2018-04-19:
@ButlerRobotics Excuse me! I would like to know if you finally succeeded. I am a novice in this area andmeet the same problem as you did.