ethz-asl/ethzasl_sensor_fusion #46: rosmake can't build

zanazakaryaie, 2017-07-19:
Hello

I installed ROS indigo on raspberry pi successfully and followed the instruction to install ethzasl_sensor_fusion (first image) but when I type "rosmake ethzas_sensor_fusion" I see that no packages has been installed! (second image)

[image]

[image]

I searched for the problem. One mentioned [the same for ethzasl_ptam (http://answers.ros.org/question/185920/problems-with-building-ptam/) and solved it by copying the download package to source folder of catkin. I did this but the same thing happened. I think ros cannot undentify the package or something like this.

Could you please give me some hints how to solve this?

helenol, 2017-07-20:
Use catkin build. http://wiki.ros.org/catkin/Tutorials/using_a_workspace

zanazakaryaie, 2017-07-28:
Dear Helen

Thanks for your reply

I used catkin_make and it worked on my raspberry pi.

Now I want to try the same in Ubuntu. I get the following error:

*CMake Error at catkin/cmake/catkin_workspace.cmake:95 (message):  This
workspace contains non-catkin packages in it, and catkin cannot build  a
non-homogeneous workspace without isolation.  Try the
'catkin_make_isolated' command instead.Call Stack (most recent call
first):  CMakeLists.txt:63 (catkin_workspace)-- Configuring incomplete,
errors occurred!See also
"/home/zana/ros_catkin_ws/build/CMakeFiles/CMakeOutput.log".See also
"/home/zana/ros_catkin_ws/build/CMakeFiles/CMakeError.log".Invoking "cmake"
failed*

could you please help me?

Regards,
Zana

ZacharyTaylor, 2017-08-02:
Dear Zana,

None of our packages are designed to work with catkin make and so it is just luck if it can actually build them. 
Did you try the catkin_make_isolated command suggested by the error?
Otherwise I suggest switching to catkin build, see here for more details http://catkin-tools.readthedocs.io/en/latest/migration.html

zanazakaryaie, 2017-08-02:
Hi

Thanks for the reply

Isn't there any ROS free version out there?